SFUSatClub/obc-firmware #9: Create Queue to Log Errors (B)

richarthurs, 2017-05-30:
When the satellite is in operation, it is inevitable that errors will occur. These errors must be logged to our nonvolatile memory, where they are saved for downlinking later. 

Right now, we don't have the memory interface working. So we need to create a queue to contain error messages that other tasks will create (such as the watchdog task, notifying us of an external reset signal). 

This queue will contain an error structure, which also needs to be defined. At minimum, it should have the following features:

- Time stamp (this will interface with the RTC later), right now it can be considered as a big number

- Some sort of error code or description. This also needs definition and documentation. 

For testing, a task that periodically looks at the error queue and returns all of the data over the UART is required. This could also be connected to the command interface, so we could issue a command to look at the logged error information.

ultrawide, 2018-03-28:
Few questions,
1. How big does the logging queue need to be to store telemetry data?
2. For documenting error codes and message types, is there a outline/resource we should refer to?
3. As per, "Create function to send the telemetry queue data out over UART", what is the UART?

liquiddandruff, 2018-03-28:
1. Lets say... 20 (unique) events for now. Arbitrary, but 20 unique errors (at once) seem hard to reach, especially if a consumer is working away at it. If there's a need we can increase this in the future.
1. There will probably be a big enum table of error codes. Anything that is an error condition/something that went wrong that we should know about will have its own enum and will be reported on the error log. We can look at any of the modules for inspiration at what can go wrong. It might also be useful to have in each event a 4 byte bitfield to log the instantaneous state of the satellite (i.e., current blocked/running state of each task, on/off state of various systems, etc).
3. Serial. It's the output you see when you connect to it over usb, via things like `serialSendQ` in the source.

ultrawide, 2018-05-20:
I followed how another task receives items from the queue in sfu_tasks.c.
```
while (xQueueReceive(xSerialTXQueue, &txCurrQueuedStr, xTicksToWait) == pdPASS) {
			serialSendln(txCurrQueuedStr);
		}
```
The intent was for a task wake up only when an item in the queue is available, rather than continuously polling. Any suggestion is appreciated!

liquiddandruff, 2018-05-20:
As per https://www.freertos.org/a00118.html we see that

> specifying the block time as portMAX_DELAY will cause the task to block indefinitely (without a timeout).

So doing the below should work: 

```
xQueueReceive(xSerialTXQueue, &txCurrQueuedStr, portMAX_DELAY)
```

commit 9d86b5c2 (2018-05-25): Create Queue to Log Errors; closes #9

* task to add items / remove items from queue
* changed MaxHeapSize to 38000

Squash merged from Colin (#65)
closed 2018-05-25 by liquiddandruff