SFUSatClub/obc-firmware #18: Test safety features of TMS570 and incorporate into system (self-checkup procedures on boot, etc)

liquiddandruff, 2017-07-19:


richarthurs, 2017-10-05:
- Enable CCM (core compare module) in HALCoGen (won’t run in debug mode)
- Get self test pass/fail information out from the tests (currently they just loop forever on failure)
- Write a test suite to inject failure in self test module, see if we can detect them
- sfu_startup.h

richarthurs, 2018-04-26:
Here are some more details:

- enable self tests at startup (halocogen, sys_startup.c, sys_selfcheck.c,  sfu_startup.c)
- once we get into our software, we should be checking the values of the self check error registers, such as FSRC0 and logging them to a file. We should check the registers for every self test that we perform. 
- TRM section 7 and 8

Beyond self tests, there is ECC memory. There are registers that count the number of corrected errors and things like that.
- enable anything required for this functionality (halcogen)
- figure out which registers to look at (TRM pg. 302 is a starting point)
- Write a function to check these registers. We will periodically run this during operation

TRM: http://www.ti.com/lit/ug/spnu607a/spnu607a.pdf

richarthurs, 2018-04-26:
This library might also have examples of how to do some of the things we're interested in: http://www.ti.com/tool/SAFETI_DIAG_LIB

slanab, 2018-07-07:
Merged to master in 5f17ecbc846132de075e18b03c943acb586ad9d6

closed 2018-07-07 by slanab