SFUSatClub/obc-firmware #25: Safe mode entry

richarthurs, 2017-08-28:
We need to decide what sort of conditions will cause us to enter SAFE (beacon mode). 

One good one might be the amount of heap space remaining decreases below a critical level. See [here, heap_4.c section](http://www.freertos.org/FreeRTOS-Coding-Standard-and-Style-Guide.html)

liquiddandruff, 2017-08-28:
That could be a possible entry condition. Before it gets to that point however, we should take care not to do dynamic allocations in user code at all, barring the one-off RTOS task creations on boot. That way we can avoid this problem entirely. 

I wonder what might come up that will warrant entry into safe mode that a power cycle from the watchdog won't fix. Maybe situations where bootup/self-checkup/peripheral initialization procedures fail repeatedly in quick succession will warrant entry into safe mode. That way we can take a look at what went wrong and try to fix it before giving it a go-ahead to try booting again. That will save power and stuff too.

richarthurs, 2018-04-21:
I'm thinking entry to safe mode will happen under the following conditions:

- no contact from ground for a long time
- sensor values out of expected range
- filesystem full
- automatically when we start up (this covers a watchdog reset too)
-