SFUSatClub/obc-firmware #27: Internal Watchdog Task

richarthurs, 2017-10-05:
We need a task to periodically tickle the internal watchdog. We also need a simple function to toggle it outside of the RTOS. 

In HALCoGEN there is an example project that uses the internal watchdog. Help > Examples > something involving DWWD (digital windowed watchdog)

mathison54, 2018-05-17:
Hi!
This sounds cool, I could try to do it.

richarthurs, 2018-05-17:
Hi Adriano, thanks for the PR and for your interest in the project!

Definitely give it a shot if you're interested. Post a comment or PR when you finish and I'll test it out on our hardware for you.

richarthurs, 2018-05-17:
I'll just note that the HALCoGen example uses a timer to trigger the watchdog. We would prefer executing the watchdog tickle inside a FreeRTOS task. You can look at the blinky function in sfusat/sfu_tasks.c if you need a simple task example. We want it in a task so that we can suspend the task from the ground to reset the system.

richarthurs, 2018-05-17:
You can put the task implementation in `sfu_tasks.c`, and you can create the task itself in `task_main.c` alongside all of the other tasks. 

Videos #1 and #3 [here](https://www.youtube.com/playlist?list=PLcguvqXCBMy1I6HJXY4a4-JYDu080Ofvw) might be helpful, although they are a little out of date.