SFUSatClub/obc-firmware #35: Low Power Proof of Concept

richarthurs, 2018-01-30:
Using tickless idle, implement a proof of concept of low power entry.

liquiddandruff, 2018-02-01:
Related #22 + note that this would mean moving to cooperative scheduling from preemptive (big change).

richarthurs, 2018-02-05:
Is that still the case? I didn't see a reference to cooperative here: [https://www.freertos.org/low-power-tickless-rtos.html](https://www.freertos.org/low-power-tickless-rtos.html)

liquiddandruff, 2018-02-06:
Ah, tickless _idle_. Nevermind you're right, the stuff about the scheduler won't be relevant here: we *do* want the option for the idle task to run without tick interrupts, and doing so won't involve any change to the current scheduler.

I misread and thought we want to change the whole RTOS to be tickless :)

richarthurs, 2018-07-07:
Background reading: 
- TMS570LS0714 technical reference manual sections: 2.4.3, all of section 3 http://www.ti.com/general/docs/litabsmultiplefilelist.tsp?literatureNumber=spnu607a
- the low power sections at the bottom of this document: https://docs.google.com/document/d/1KGF3oE50fsP7u-onwebT7V9FUQS2UwFoQvcsV4_4IAY/edit

When searching for forum posts, try things like "hercules low power", "TMS570 sleep"

Power Saving Methods
- disable power domain 5 (we don't need it and will never need it)
- idle task hook entry into low power mode: https://www.freertos.org/RTOS-idle-task.html
- tickless idle: https://www.freertos.org/low-power-tickless-rtos.html

## Idle Task Low Power
The idle task is run by the RTOS at the lowest priority, and is used for freeing up memory and things like that. Using the idle task hook, which is just a function that the idle task will call, we can enter into a low power mode. When this is done, it means we are very frequently entering a low power mode for the duration of an RTOS tick (1 ms), and then exiting to check if other tasks need to be executed. Despite the frequent switching, it can still result in power savings. 

To use this, we must place the processor into a sleep mode that is able to be woken up by the RTI (timer) interrupt, which is the RTOS tick. 

This post has mention of keeping the RTI going while in low power mode: https://e2e.ti.com/support/microcontrollers/hercules/f/312/t/640585?CCS-Wakeup-hercules-from-Snooze-mode

This post has details about how interrupts are used to wake up from sleep, and a really simple sample application:
https://e2e.ti.com/support/microcontrollers/hercules/f/312/p/640687/2363875#2363875

We can essentially take the sleep() function from the post above and use that as our idle task hook, as long as everything is set up to wake up on RTI interrupt. 

## Tickless Idle
Tickless idle seems to be what we think about as “low power state” - it is controlled by the application and won’t happen automatically.  https://www.freertos.org/low-power-tickless-rtos.html 

Tickless idle will keep the system tick from interrupting the idle task constantly. To do this, it will do the following, when we ask to enter tickless idle:
- Double check that it’s ok to stop running the RTOS (no tasks need to be executed)
- Start another tick that is operable in a low power mode. This will time how long we’re in the low power mode. Call this the low power tick. 
- Enter low power mode
- Exit from the low power mode is from some sort of interrupt - either a low power timer, an external event (such as RTC alarm or communication over radio)
- When it exits, using the low power tick, it will determine how long it was sleeping for. The system tick will be adjusted accordingly. Tasks will restart.

This is the second thing we'll do. Low power state will place the satellite into low power mode for an extended period of time (multiple seconds), and will be woken up by some external source, such as an alarm from the RTC, or an incoming radio message. To get this working, we will need to set up the low power timer, and change some application logic to suspend everything before entry into this mode. It will be entirely controlled by us. 

## Experiment 1
Using a launchpad and by adapting the code from here: https://e2e.ti.com/support/microcontrollers/hercules/f/312/p/640687/2363875#2363875

- Do not call sleep() at the top of main. You MUST have a long busy-wait loop (multiple seconds) at the top of main. Once the wait is done, you can sleep. There have been forum posts about people bricking their chips by having them sleep as soon as they get power, which makes them impossible to program. 
- Figure out what registers are being written to during the sleep function
- Figure out if any wake up source is enabled with the code
- Create a demo that allows us to enter and exit sleep mode with a GPIO

## Experiment 2
Implement idle task hook sleep mode. The wake up sources will need to be changed for this, but if experiment 1 is successful, experiment 2 should be straightforward.