SFUSatClub/obc-firmware #39: Single branch supporting multiple platforms

liquiddandruff, 2018-01-31:
Currently things are split up in branches (hercules launchpad in branch **main**; OBC 0.4, OBC PROTOTYPE A, and another hercules launchpad in branch **dual_config**). The goal is to not need to switch branches should if one wants to develop/test on another platform. 

In branch dual_config, sources in ./include and ./source seem only suitable for the platform in active development (in this case, OBC 0.4). Since we can't easily rely on modifying these files (since they are regenerated via HALCoGen), we won't be able to do the usual of adding include guards to properly support the project configurations currently listed (i.e., LAUNCHPAD project configuration likely won't work in this state short of regenerating all ./include and ./source sources -- correct me if i'm wrong).

Proposal: The single branch will have multiple platform-specific ./<platform>include and ./<platform>source directories that will be selected appropriately based on the chosen CCS project configuration. Our actual source files in ./SFUsat will then sport platform-specific ifdefs as necessary to build and run on the appropriate project configurations.

richarthurs, 2018-01-31:
Great idea, that'll make it much easier. I think just moving the halcogen files into another directory (one per config) should have it generate the include and source directories under that one: 
/config_name/source
/config_name/include

Base this change off the current dual_config branch since it's the most mature. Then we'll turn it into master. I think there's just some extra stuff that Seleena wrote hanging around - plus the radio branch which we can bring in later.

liquiddandruff, 2018-02-01:
Big if true (hehe). That'll make things neater too, and doubly so if it's something halcogen can play nice with.

closed 2018-02-20 by richarthurs