SFUSatClub/obc-firmware #42: Printf or similar

richarthurs, 2018-02-07:
It'd be nice to easily print strings that include printf-style format specifiers using the UART.

Printf is heavy, but there are several light weight implementations. 

https://mcuoneclipse.com/2014/08/17/xformat-a-lightweight-printf-and-sprintf-alternative/

richarthurs, 2018-03-02:
I set this up: https://github.com/mpaland/printf - lightweight and reentrant, will be pretty useful for testing

sprintf works, still need to test printf and make rtos version.

liquiddandruff, 2018-03-03:
Nice, that one looks good.

richarthurs, 2018-04-15:
Printf, snprintf added by: ec9f11d19621a03495d80d11c12893efc1503844

Added [wiki page](https://github.com/SFUSatClub/obc-firmware/wiki/Printing-to-the-Terminal) detailing appropriate use cases for various functions. 

Printf uses a lot of memory, but it can sometimes be helpful to enable it (debugging SPIFFS, where printf verbose mode can be enabled, for example).

closed 2018-04-15 by richarthurs