SFUSatClub/obc-firmware #46: Interrupt Driven Packet Radio

TobiNakamura, 2018-02-26:
The following radio CC1101 events should be handled
1. Packet transmit complete. If it does not transmit, try to fix it.
2. Packet received and CRC checked. Read RX FIFO and handle data appropriately.

TobiNakamura, 2018-05-13:
RX interrupt GDO should be configure to 0x01: "Asserts when RX FIFO is filled at or above the RX FIFO threshold or the end of packet is reached. De-asserts when the RX FIFO is empty."
RX FIFO threshold is set by 0x03: FIFOTHR.FIFO_THR = 0b1111

The ISR should immediately take data from the RX_FIFO and then check CRC status.

nick1au, 2018-06-24:
Issue 1 regarding this task: Interrupts are being triggered incorrectly: GD01_CFG with 0x01 results in an immediate interrupt despite threshold being set to 30, 60 and 64 bytes on RX FIFO and being in RX state. 0x07 does not interrupt at all (CRC was OK).
Issue 2: The system was producing an interrupt upon startup causing reading of registers to be incomplete despite being connected to no pins at all.
Issue 3: Once CC1101 interrupts once and returns to normal state, the Launchpad no longer interrupts as intended and runs in a suspended state.

Issue occurs on RF Rev2 board with label 2 and Launchpad with no labelling

Due to time constraints, basic functions for RX and TX are being used in a loop with delay.

richarthurs, 2018-07-22:
Pretty sure we can close this eh? @liquiddandruff

closed 2018-07-22 by liquiddandruff