SFUSatClub/obc-firmware #47: Infinite Packet Length

TobiNakamura, 2018-02-26:
CHIME calibration signal will use the infinite packet mode to continuously transmit arbitrary lengths of Gold's Code.
Current implementation only supports fixed length packet mode with packet length < FIFO length.

TobiNakamura, 2018-05-13:
See section 2.3 of [DN500](http://www.ti.com/lit/an/swra109c/swra109c.pdf) and the infiniteLink example.

1. Set PKTCTRL0.LENGTH_CONFIG = 2 (10) (Infinite packet length).
2. Set FIFOTHR.FIFO_THR = 0b1110
3. Set GDO.CFG = 0x02
3.  Pre-program the PKTLEN register to 0.
4. Transmit 64 bytes
5. repeat
5.A on falling edge interrupt load 59 bytes to FIFO
5.B reattach interrupt
6. Set PKTCTRL0.LENGTH_CONFIG = 0 (00) (Fixed packet length) to end transmission

Interrupt:
GDO is set to mode 2 which will inform the OBC to top up the TX FIFO. GDO transitions to low when the contents of the FIFO falls below the number of bytes specified in register 0x03: FIFOTHR.

richarthurs, 2018-07-25:
This might be a good candidate for a FreeRTOS stream buffer: https://www.freertos.org/RTOS-stream-message-buffers.html (just a thought, I don't know much about them)

They're not available until v10 (we're currently on v9), but upgrading is fairly easy and we'll upgrade for CSDC5/CCP.