SFUSatClub/obc-firmware #48: Linker Command File is Old

richarthurs, 2018-02-27:
I was seeing our RAM usage approach 70% ... which seems too high. Upon further inspection, the linker command files are identical between launchpad, 0.3, 0.4 configs. This is bad since the launchpad uses a really wussy chip, so the address spaces are much smaller than on the 0714. 

Todo: generate a new project with halcogen for 0714PGE + RTOS, grab that linker cmd file. 

It's weird that this isn't re-generated by halcogen...

liquiddandruff, 2018-02-27:
This was one of the first things I noticed too, but haven't investigated it further than opening up the memory map view and finding nothing amiss. Nice catch with the linker file, it seems likely an issue with the addresses used then.

richarthurs, 2018-02-28:
fixed in 841a5637ee06549431b0ec2b56504ce7d53424f9

closed 2018-02-28 by richarthurs