SFUSatClub/obc-firmware #53: Move as much from sys_main.c to sfu_task_main.c as we can

liquiddandruff, 2018-04-03:
Each platform has their own `sys_main.c`, duplicated with the same code.

We should see if we can move this initialization immediately inside `vMainTask`, before any other task is created (so before `vMainTask` can be pre-empted). I believe this has been attempted before, but due to reasons (e.g., desires to print to serial on bootup/MPU permission errors preventing certain hardware initialization to work when in a FreeRTOS task), necessitated adding these initialization to the `sys_main.c`. 

Let's revisit this and see what can be improved.

richarthurs, 2018-04-15:
@liquiddandruff I think we can close this now, eh?

liquiddandruff, 2018-04-15:
That's right, thanks

closed 2018-04-15 by liquiddandruff