SFUSatClub/obc-firmware #57: Define TASK_IDs as unique magical identifiers instead of consecutive enums as they are now

liquiddandruff, 2018-04-22:
Much like the move from cmd ids enums to magic identifers as in commit c98310ec72fc9e

mathison54, 2018-05-16:
Hi,
I could work on this if you like. So just needed to change it from enums to defines?
About TASK_SERIAL, TASK_BLINKY and TASK_NUM_IDS, I would put them under #idfedef, could it be "_DEBUG"?

liquiddandruff, 2018-05-16:
Hi Adriano, we'd definitely appreciate your help! Yes, from enums to defines similar to the commit linked. 

Regarding the two task IDs originally meant for debug (TASK_NUM_IDS is a count and not a task id), TASK_SERIAL is to be used in release as well now, and there may be things depending on TASK_BLINKY, so just copy things over as is and update the comment (don't bother with ifdef).

closed 2018-05-19 by liquiddandruff