SFUSatClub/obc-firmware #60: Sun Sensor Multiplexer Function

richarthurs, 2018-05-04:
We have several sun sensors which are multiplexed to 4 ADCs. We have 16 sun sensors, and we need to get values from all of them. 

- each multiplexer switches 4 sensors into a single ADC (if I remember right)
- ADCs are read in bulk (there are several channels on an ADC, we use 4 channels for the sun sensors, one for each multiplexer)
- so we need to set all muxes to channel 0, read the ADCs, record the 4 values, set muxes to channel 1, read the ADCs for the next 4 values, and so on 

This requires:

1. a function to send a command to a particular multiplexer to set it to a particular channel
2. a function to read an ADC (we have this in sfu_tasks.c, vADCRead)
3. a function to loop through and set the multiplexers up, then read the ADC values (it will call the above two functions). This will read every sensor. 

# Step 1:
- Create a new branch off richard_temp&filesys (this has stlm75.h which is an I2C example)

- Create a new header and .c file (sun_sensor.h). 

- Using the multiplexer data sheet: https://ca.mouser.com/datasheet/2/609/ADG728_729-878514.pdf, and the information in https://github.com/SFUSatClub/obc-firmware/wiki/I2C, create the function to set a multiplexer's channel

# Step 2:
Don't worry about the ADC for now, we'll deal with it later. 

# Step 3:
Create the 3rd function. We will need some sort of look up table to map the multiplexer number and ADC channel to the sun sensor number. Start with dummy values. We need to be able to loop through this table and grab readings from each mux/ADC combination (each sensor)

Table example: we need to loop through this 

Mux   |   ADC   |   Sun Sensor
  0      |     12     |            0
  0      |     13     |            1
  0      |     14     |            2
  0      |     15     |            3
  1       |     12     |            4
  1       |     13     |            5
  1       |     14     |            6
  1       |     15     |            7
  2       |     12     |            8
  2       |     13     |            9
  2       |     14     |            10
  2       |     15     |            11
  3       |     12     |            12
  3       |     13     |            13
  3       |     14     |            14
  3       |     15     |            15

TobiNakamura, 2018-05-04:
Although each mux has 8 inputs not all are used! The number of sun sensors per mux is dependent on the hardware configuration which can be anywhere from 4 to 6. Thus the need for a look-up table of the form:
Sun Sensor | Mux | Mux Pin | ADC
0                  | 0      | 1            | 0
1                  | 0      | 3            | 0
2                  | 0      | 6            | 0
3                  | 0      | 7            | 0
4                  | 1      | 1            | 1
5                  | 1      | 2            | 1
6                  | 1      | 3            | 1
7                  | 1      | 5            | 1
8                  | 1      | 6            | 1
9                  | 1      | 7            | 1
etc.

Note that a particular mux will always have the same ADC since that is not configurable in hardware.

x+ mux - address 0x4C
            -  output goes to PC/104 26B
x- mux - address 0x4D
            -  output goes to PC/104 24B
y+ mux - address 0x4E
            -  output goes to PC/104 23B
y- mux - address 0x4F
            -  output goes to PC/104 25B

richarthurs, 2018-05-04:
Thanks for the clarification!