SFUSatClub/obc-firmware #64: CC1101 Register Read issue

TobiNakamura, 2018-05-20:
Register 0x00 indicates value mismatch upon loading and then checking the register value.

The problem has now shifted to register 0x02. Showing 0x1e as the stored value instead of 0x06.

liquiddandruff, 2018-05-20:
As a note, here was what I got last time with the below code:

[image]

Output:
```
Radio Status Registers:
00, 14, f7, 00, b4, 0d, 00, 0
#2 reg=0x00 0x00!=0x5c
radio registers do not match!
```

Desired values:
[image]

nick1au, 2018-05-20:
[spi log.txt](https://github.com/SFUSatClub/obc-firmware/files/2019970/spi.log.txt)

Here's the log from the serial console and the logic analyzer in SPI mode (MISO and MOSI). Issue was reproducible in both console and analyzer.

Radio Status Registers:
f0, 0f, 1e, 0f, 3c, 0f, 1e, 0
Reg 02 = 1e != 06
radio registers do not match!

0.108540000000000,16,0x02,0x0E - register 0x02 (iocfg0) write. status indicates that it is in idle
0.108549625000000,16,0x06,0x1E - data value 0x06 written to reg2. status indicates RX mode
0.208546250000000,17,0x01,0x0F - remaining is in idle state
0.208556000000000,17,0x2E,0x00
0.308552750000000,18,0x00,0x0F
0.308562375000000,18,0x5C,0x00
0.408559375000000,19,0x03,0x0F
0.408569125000000,19,0x40,0x03

liquiddandruff, 2018-06-12:
Issue has been identified. Long story short, some entries in the RF configuration arrays (i.e., `SMARTRF_VALS_TX`) where removed, causing writes to addr 0x00 with val 0x00 on each instance of missing entry. Fix is to repopulate these entries (like the missing `SYNC` entry: `[SYNC1] = SMARTRF_SETTING_SYNC1_VAL_TX,` etc) and add compile/run-time check that there are no values unset.

nick1au, 2018-06-24:
Something to note with this issue: one of the RF boards (June 2017 Rev2 with two coax connectors and no number labels) has issues with read/write to multiple registers as they are not written correctly (see below). This was not occurring on another Rev2 (labelled #2 with one coax connector), and swapping Launchpads and boards from a known good setup produced the issue to this specific RF board.

Register check changes made on commit 7b71a0cbcc028fe949e35d8bee2f6fe19668eb6e (2018-06-23)

Radio Status Registers:
00, 14, 00, 0f, 80, 01, 00, 00
Reg 02 = ff != 06
Reg 00 = 29 != 5c
Reg 03 = 07 != 47
Reg 06 = ff != 1e
Reg 08 = 45 != 04
Reg 0b = 0f != 06
Reg 0d = 1e != 10
Reg 0e = c4 != bb
Reg 0f = ec != 13
Reg 10 = 8c != f5
Reg 11 = 22 != 83
Reg 12 = 02 != 03
Reg 15 = 47 != 13
Reg 18 = 04 != 18
Reg 19 = 76 != 16
Reg 20 = f8 != fb
Reg 23 = a9 != e9
Reg 24 = 0a != 2a
Reg 25 = 20 != 00
Reg 26 = 0d != 1f
Reg 2c = 88 != 81
Reg 2d = 31 != 35
Reg 2e = 0b != 09
radio registers do not match!

commit 84ac9725 (2018-06-15): Fixes #64