SFUSatClub/obc-firmware #67: Standard Telemetry

richarthurs, 2018-05-23:
Standard telemetry refers to the telemetry packets we constantly send out. 

These have all of the basic health info for the satellite, and we always send out the most recent values. 

I propose the following tasks:
- `stdtelem_send`: sends the most recent stdtelem packet out on the radio
- `stdtelem_collect`: runs the various checker functions to log telemetry

The reason for two tasks is that the sending of telemetry and the acquisition of telemetry are distinct. We may not always want to send out telemetry (to save power), but we definitely always want to be logging it. 

**stdtelem_collect**
This task runs various functions to log telemetry. Different telemetry points are logged at different rates. Checker functions will log errors using the error log system if telemetry points are out of expected range. 

To support some ground commands for configurability, we need:
- data structure for max and min telemetry reading ranges
- data structure for telemetry point logging frequencies
I'm thinking a straightforward enum - this would make it easy to write a command to tweak the max or min for a specific telemetry point, for example. 

A telemetry logging function shall:
- run if it is the correct time (so it logs at the correct period)
- read the telemetry from the sensor
- (if applicable) compare with the max/min. Log an error if outside of range, and potentially trigger entry to safe mode if outside of range
- save the telemetry to the appropriate log
- update the telemetry point so that the `stdtelem_send` has the recent value

An issue up for debate is whether we should have separate tasks for each telemetry type. It would make it easy to implement the timing, but would result in many tasks (5-10 additional). However, it would allow us to prioritize logging certain telemetry items (although we never log anything super fast, so we probably don't need to worry about this). 

Telemetry points:
- battery state of charge + other things
- OBC current draw (and some others?)
- temperatures (5 or 6 across the sat)
- sun sensors
- number of errors logged
- filesystem remaining space + other things

richarthurs, 2018-05-23:
@liquiddandruff do you agree with this plan? And what's your opinion on 1 vs. many tasks to do the logging?

liquiddandruff, 2018-05-24:
Good plan, and yes many tasks for this makes sense.

richarthurs, 2018-05-24:
Cool beans, I'll start it up

richarthurs, 2018-06-27:
We have a bunch of stdtelem now

closed 2018-06-27 by richarthurs