SFUSatClub/obc-firmware #69: Add logging for comments such as "todo: log here"

ultrawide, 2018-06-02:
Look for calls such as `// todo: log error, i2c was reset` and add in respective `addLogItem()`calls.

ultrawide, 2018-06-02:
Here's a list of where I found cases to log items in /sfusat and /SPIFFS

/sfusat/sfu_i2c.c line 156
/sfusat/sfu_task_radio.c line 262
/sfusat/stlm75.c line 53  
/SPIFFS/sfu_fs_structure.c line 406

How should I structure the enum for log types? Where should we create a reference that details what each log item means?

richarthurs, 2018-06-05:
You can make an enumeration entry for the driver type. So one for filesystem, one for i2c, etc. How we split things up isn’t too critical, since we can always look back at the code to see where things came from. We’re not automatically acting on the log entries. 

Please make a google sheets document in the computing folder on the drive. Also add a link to it in the wiki entry for this code.

ultrawide, 2018-06-06:
Created the reference document. Also updated on the github wiki: error logging.

https://docs.google.com/document/d/1VdZ2ZLdfByL5thwYg4ggyNYvPFjmcHge-LwsazGSZjg/edit

closed 2018-06-06 by ultrawide